AlexTelon/FlexiCharge-BMS #67: [Task]: Create communication protocol specifications

ViktorM02, 2023-09-21:
### Parent Backlog Item

#38

### Description of new task

We have the base document for the communication protocol specifications between the charger and bms, now we should add some basic content, such as:

* The handshake .
* What data will be sent from the bms?
* What data will be sent from the charger system?
* Heartbeats? How often?
* The rate the data should be sent? Should it be sent with a fixed frequency?

### Required Manual Steps

- [X] Confirm title is filled out
- [X] Additional labels added.
- [X] Confirm parent backlog item has correct issue id

KevFord, 2023-09-24:
Started working on writing up a basic communications protocol. Need to find out how often the heartbeat and data updates should happen.

KevFord, 2023-09-28:
Approved and merged with dev.