AlexTelon/FlexiCharge-BMS #103: Establish and test connection with charger

ljungqvistaxel, 2023-10-06:
To be able to seamlessly charge the battery, the charger and BMS need to communicate according to specification document. This meaning that every message type is tested for functionality with both systems connected.

Requirements
- Every message type that can be sent over the UART wires should be tested together with the charger connected.
- A full charging session (from chargers perspective) should be able to take place without human interaction in the BMS

ljungqvistaxel, 2023-10-06:
Effort: 5

KevFord, 2023-10-11:
The software has been tested with both squad's hardware and it is working as intended. Software is currently awaiting review.

KevFord, 2023-10-11:
Closed with the completion of task #98 and #103.

closed 2023-10-11 by KevFord